Ultimaker/Cura #20892: Enable Support Interface button grayed out

agunievass, 2025-08-24:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Hellbot Manga 2 230

### Reproduction steps

1. Check the "Enable Support Interface" button
2. One or both (Roof, Floor) will be selected
3. Select only "Enable Support Roof"
4. Deselect "Enable Support Roof"
5. The submenu closes and the button "Enable Support Interface" is greyed out, cannot be activated again, a message saying "This setting is not used because all the settings that it influences are overriden". Making it imposible to activate again.

### Actual results

The button "Enable Support Interface" is greyed out, cannot be activated again. Even when checking and unchecking the settings in "Settings visibility"

### Expected results

The menu should close, but not be greyed out, so it is available to be activated again. When pressed again the Floor and Roof should be activated by default.

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2025-08-25:
Thanks for the report.
5.6 is 7 or 8 versions behind the current version.  I don't have a problem with 5.10.1.

I don't see any changes to the fdmprinter.def.json file for that setting.  Have you tried using the visibility tool (the hamburger button next to the Search Box) to insure that they are visible.

Visibility tool.

[image]

Settings
[image]

agunievass, 2025-08-25:
Thanks for the reply, i now updated to Cura 5.10.2 (latest) and im still able to reproduce the error, but now i add a clarification:

1st Select a printer profile having "Enable support interface" turned off by default:
[image]
2nd Enable the support interface, since you made changes to the profile a floppy disk button appears on top (giving the option to save the changes, but we dont need to) and a circular arrow appears by the side of "Enable support interface"
[image]
3rd Now uncheck the "Enable support roof" and then hit the blue circular arrow by the side of "Enable support interface"
[image]
4th Now the button becomes greyed out and cannot be activated again. An option to solve this is to click the circular blue arrow on the side of the floppy disk (reseting all the changes) but this resets all the changes made to the profile; wich is a bit annoying if made a lot of changes.
[image]
The setting visibility is not an issue
[image]

The expected behavior must be that, when clicking on the blue arrow by the side of "Enable support interface" (reseting this to the default state) this setting and _all of the submenu settings_ are reset to the default state.

GregValiant, 2025-08-26:
Doing all of that, I can duplicate this.
When I re-select the Custom Profile so Cura goes to those particular defaults, the box is re-enabled and the problem goes away.

It appears that the "Circle Arrow" button might need some work.  Someone from the Cura team will take a look.