Ultimaker/Cura #20898: Printbed goes down too far

Eliteronin319, 2025-08-27:
### Cura Version

5.9.1

### Operating System

windows 10

### Printer

Tronxy x5sa pro

### Reproduction steps

1) I canceled any print (happens every time atm, wasn't an issue some weeks ago. Someone on a forum mentioned some probable failure in the gcode. So im sending the actual g-code from cura itself and the file after it was sliced using an Editor.)

[image]
[image]

### Actual results

The print bed went down to the bottom of the printer and then tried to go further, resulting in a screeching sound. 
I couldn't move the bed until it happened.

### Expected results

The printer should just cancel my print

### Add your .zip and screenshots here ⬇️

[image]
[image]

GregValiant, 2025-08-27:
Thanks for the report.
I saw your post on the Forum.  I don't see a problem in the Gcode.  The "Ending Gcode" might be a problem but that comes from the printer definition rather than from Cura.  The Cura portion of the Gcode ends with the Z at whatever the Z height of the final layer was.

The author of the TronXY definition files appears to be ""KV/AdderMk2" who was either contracted with by TronXY or works for TronXY.  UM doesn't (can't) get involved in other manufacturers printer definition files.

As PartySausage and Slashee pointed out on the Forum, "two commands on one line" is a problem.  There may also be something going on when the M84 line tells the steppers to shut off.  Some printers have been known to travel down all by themselves if the Z motor is disabled.

I've gone over the stock Ending Gcode and made some changes.
Copy and paste this into the "Ending G-Code" box in the Machine Settings in Cura.

;----------Ending Gcode
G92 E0 ; Reset Extruder
G91 ; Set Positioning to Relative
M83 ; Set Extruder to Relative
G1 E-{retraction_amount} F{retraction_retract_speed*60} ; Retract
G1 F{speed_z_hop*60} Z0.2 ; Raise nozzle .2mm
G1 F{speed_travel*60} X5 Y5 ; Wipe out
G1 F{speed_z_hop*60} Z10 ; Raise nozzle 10mm
G90 ; Set Positioning to Absolute
G1 Y{machine_depth} ; Park print head
M82 ; Set Extruder to Absolute
M106 S0 ; Turn off Layer Cooling Fan
M104 S0 ; Turn off Hot End
M140 S0 ; Turn off Bed
M84 X Y E ; Disable all Steppers except Z
;----------End of End

I did make a change to the G1 X{machine_width} Y{machine_depth} line.  I dropped the X location so the nozzle will just move to the back and remain at wherever it was after the last move of the print.  It won't have as far to go for the next print.

See if that works for you and let me know.  I really don't see a Cura problem here.  The file would print correctly on my Ender (once the StartUp and Endings were adjusted for my machine).  You should also take Party Sausage's advice and check the Z stop switch.  It should not allow the printer to go to a negative Z location.

HellAholic, 2025-08-29:
@GregValiant I added a PR to fix the typo (missing \n), that is part of the printer-linter.

GregValiant, 2025-08-29:
More than one of the Tronxy definitions could use some work.
That snippet I posted would be generic.  (But just because I like it doesn't mean other folks will.)