Ultimaker/Cura #20905: Infill Travel Optimization causes slicing to never finish on relatively simple part

gjsmo, 2025-08-29:
### Cura Version

5.10.2

### Operating System

Windows 10 Enterprise 22H2

### Printer

Voron2 350

### Reproduction steps

1. Load attached 3MF
2. Slice

### Actual results

Slicing continues for at least 30 minutes, stalling around 60% after less than a minute.

### Expected results

Slicing should complete in a reasonable amount of time. With Infill Travel Optimization disabled, slicing completes in a little over one minute.

### Add your .zip and screenshots here ⬇️

[V2_ASA_GRIPS_-_620x525_(4).zip](https://github.com/user-attachments/files/22054581/V2_ASA_GRIPS_-_620x525_.4.zip)

[cura.log](https://github.com/user-attachments/files/22054665/cura.log)

GregValiant, 2025-08-30:
Thanks for the report.
The repair utilities I use didn't find any errors in the model.  I turned off "Infill Travel Optimization" and it took about 1 minute to slice.  With Infill Travel Optimization turned on...it took a really long time.

This happens to be from PrusaSlicer.  Every one of the little "cup" shapes looks like this:

[image]

when the model looks like this:

[image]

Anyway, Slicing finally finished.  From my log file (with Infill Optimization turned on):
_"Slicing took 3127.25 seconds"_.  For those of us who are not Dr. Spock - that's  52min and 7 sec.
Compared to with Infill Optimization turned off:
_"Slicing took 60.03 seconds"_ which is about 1 minute and 2 sec.

@HellAholic will take a look.

This is the Cura preview.

[image]

HellAholic, 2025-08-31:
The Infill Travel Optimization option is experimental meaning there are known drawbacks when the setting is enabled, and as noted in its description, it can significantly increase slicing times depending on the model geometry.
I don't understand what you're reporting as an issue here.

```When enabled, the order in which the infill lines are printed is optimized to reduce the distance travelled. The reduction in travel time achieved very much depends on the model being sliced, infill pattern, density, etc. Note that, for some models that have many small areas of infill, the time to slice the model may be greatly increased.```

[image]

GregValiant, 2025-08-31:
@gjsmo The Cura team is aware of the long slice times.  It would appear that for now, this is expected behavior.  The setting may be worked on in the future.  In the meantime, for certain models - the fix is to turn the setting off.
I'll remove the bug label and close this.  I hope you understand.

closed 2025-08-31 by GregValiant

GregValiant, 2025-08-31:
The setting may have been worked on between the versions.  I think the UM preference is to move the settings out of "Experimental" and into a real category.  They need feedback (and this report qualifies) so when someone at UM has a light bulb go off in their head, then they can fix the code or add something, or even remove it all together (like "wire printing" went away).

I would not characterize that model as "simple".  There is a lot going on, and it all happens within a few layers.  The only place to "comb" within the model are those "bars" (mullions?) between the other features.