Ultimaker/Cura #20919: slicing fails sporadicaly on watertight models also

partach-beep, 2025-09-04:
### Cura Version

5.10.0

### Operating System

Windows 11

### Printer

Creality Ender 3 / Ender-3 v2

### Describe your problem and add the package for technical support as a .zip here ⬇️

Some models failed to slice. Sometimes Mesh Tools shows the "not watertight" warning, when sliceing fails, but not every time!
Already updated to 5.10.1, but issue still persists.

[export-20250904-115442.zip](https://github.com/user-attachments/files/22136109/export-20250904-115442.zip)

GregValiant, 2025-09-04:
Thanks for the report.
This one showed the "not watertight" warning when I opened it.  It sliced OK, but I would not trust it to print correctly.
I opened it in MS 3D Builder, and it indicated errors.  When I checked it in MeshMixer there were no errors.
That sort of confusion doesn't happen often but I've learned that when the repair utilities don't agree that there are "unspecified" problems in the model.

[image]

The Cura team will take a look.  In the meantime you can move the model around on the Cura build plate.  Sometimes that will free up the bug.

GregValiant, 2025-09-04:
An on-line service I used came up with this report:
-> Analyzing file
--> 0 Naked edges  (?)
--> 0 Planar holes  (?)
--> 0 Non-Planar holes  (?)
**--> 242 Non-Manifold edges  (?)
--> 384 Inverted normals  (?)**
--> 0 Duplicate faces  (?)
--> 0 Degenerate faces  (?)
--> 0 Disjoint shells  (?)
-> Repairing: 100.00%
-> Converting back
-> File repaired
Time needed for repair (ms): 9400
**--> Vertex count changed from 37715 to 40818 (+3103)
--> Triangle count changed from 75914 to 82364 (+6450)**

So maybe MS3D Builder was right after all.

[export-20250904-115442_fixed.zip](https://github.com/user-attachments/files/22142583/export-20250904-115442_fixed.zip)

partach-beep, 2025-09-18:
Honestly, I often also ignore "not watertight" - I assumed that I don't want it to be water-tight anyway.
And totally agree: most models have errors, but do I really need 3 tools to have that checked?..
As the cura-plugin"mesh-tool" isn't that good:
I'm using Blender to create my model (-simply ones, but mine ;) ), but can't find the errors in my models easily. -> Can you link me to the correct tool/setting to find errors easily? (I know the "Face Orientation"... but surely that's not all).
Or do you recommend simply using the auto-fixing with e.g., MS 3D-viewer?