Ultimaker/Cura #20925: UltiMaker Cura Version 5.10.1 : does not release Plate for Print Abort

adfile, 2025-09-05:
### Cura Version

5.10.1

### Operating System

Windows 11

### Printer

Creality Ender-3 S1

### Reproduction steps

Use USB C-Type to connect PC to Printer.
Open Cura
Select New Project
Open an stl file .
Click Slice
Select Print using USB

Once the printer begins printing, Click on Abort Print in the Cura Monitor.

After clicking on Abort Print and Cura has moved the Head and Plate to final position, it does not release the Plate.
To release the plate, it requires turning off / on the printer.
  
Note : 
Cura correctly frees the plate/head when a print completes successfully. 
 

### Actual results

After clicking on Abort Print and Cura has moved the Head and Plate to final position, it does not release the Plate.
To release the plate, it requires turning off / on the printer.

(Try moving the plate by hand .  It cannot be moved).

### Expected results

Cura needs to release the plate / head for Abort Print just as it does for successful completion.

### Add your .zip and screenshots here ⬇️

N/A

GregValiant, 2025-09-05:
Thanks for the report.
USB Printing is a legacy feature included with Cura for UltiMaker Original printers that require it.  It often does not work with other printers.

That being said, within the abort code of USB Printing is the line:
**self._sendCommand("M84")**
which you may recognize as the Marlin _"Disarm Steppers"_ command.  It is the same command used in your Ending Gcode.  
From your definition file:
**M84 X Y E ;Disable all steppers but Z**
Whether your printer is receiving it, or whether it is enabled in your firmware - we can't tell.  

A workaround might be to send the command via the "Send Gcode Command" box (I'm guessing at the name) on the Monitor page in Cura.

I'll mark this as a duplicate and close it as USB Printing is always questionable.  That is why UM abandoned it's use.  I hope you understand.

closed 2025-09-05 by GregValiant

adfile, 2025-09-05:
Typing M84 in the Monitor Send G-Code Field then pressing <Enter> frees the platter.
Thank you for the solution.