Ultimaker/Cura #20960: slicing failed with a model the prusa slicer does well with

luis1998mestre-code, 2025-09-15:
### Cura Version

5.4.0

### Operating System

windows 11

### Printer

ender 5

### Describe your problem and add the package for technical support as a .zip here ⬇️

the slicing works if I make the model a little bigger so it might be some features that could be too small but I tried changing these and it still does not work and once again the prusa slicer works ok

luis1998mestre-code, 2025-09-15:
apologies, here is the project zip file

[CE5_cage_mount_x_y_lens_flexure_fixed.zip](https://github.com/user-attachments/files/22351445/CE5_cage_mount_x_y_lens_flexure_fixed.zip)

GregValiant, 2025-09-16:
Thanks for the report.
Cura versions 5.4 thru 5.6 suffered from slicing issues on two types of models:  Organic shapes and models with a lot of holes on the initial layer.  Your model falls under the second category.
I didn't have any trouble slicing it in 5.10.2.  I suggest you upgrade.

I'll go ahead and mark this as a duplicate report and close it as 5.4 is about 10 versions out of date.

This is in 5.10.2.
[image]

closed 2025-09-16 by GregValiant